RobotLocomotion/LabelFusion #2: openni2-camera-lcm

YuehChuan, 2017-07-16:
Hi!  How could I use openni2-camera-lcm with kinet xbox360 with LCM? already install LCM and libopenni2
and I make the openni2-camera-lcm repo succeed. Type this and 
`./openni2-camera-lcm -j -z -c OPENNI_FRAME`  
[image]
most people don't know LCM and libbot, pods build system; I think had better mention those in repository.  thank you in advance!

patmarion, 2017-07-16:
Try just `openni2-camera-lcm` without any extra args.  If you look at the error output, the command line isn't parsing because you didn't pass parameters correctly.  If you really want to pass values on the command line then it would look something like:

`openni2-camera-lcm -j 90 -z 1 -c OPENNI_FRAME`

By the way, how did you end up with `~/LabelFusion/openni2-camera-lcm`?  The typical usage is to build `LabelFusion` with the `spartan` superbuild (but this under construction currently...) or the docker image which already provides a complete build.

YuehChuan, 2017-07-17:
I ever typed openni2-camera-lcm, then it showed command not found ; maybe I should put openni2-camera-lcm executable under /usr/bin. For me is docker image, forgot to run docker in the terminal.  And  I git clone  openni2-camera-lcm repo under LabelFusion folder and `make`. Thanks for notice. I'll try then update here.

patmarion, 2017-07-17:
It says `command not found` because the executable isn't under your `PATH`.  In the docker image it will be in your `PATH` already.

I recommend using the docker image, but please try it tomorrow, right now it is still under construction.
If you'd like to do the build and use `LabelFusion` without docker, these are the build steps required:

https://github.com/RobotLocomotion/LabelFusion/blob/master/docker/compile_all.sh

As you'll see in that script, all the dependencies (except ElasticFusion) are built for you when you run the Director cmake build.

YuehChuan, 2017-07-18:
@patmarion Thank you!  PATH problem fix,  and I am using docker now.
https://asciinema.org/a/7XwkqUu7MOhsnEIKlAKCfT2nJ
But it is the old docker image so it seems not work also,  I'll try the new one.

YuehChuan, 2017-07-19:
Iremove the old docker and did git pull
issue remain,  I guess is something wrong to  USB hardware reading, or I didn't setup the docker image properly.

patmarion, 2017-07-19:
I'm sorry the camera driver isn't working.  Have you been able to use this device with openni2 on this computer using another program, such as those bundled with ROS or PCL?

YuehChuan, 2017-07-19:
Yes, I could grab PCL1.8.0 point cloud, Let me check it, thanks!

peteflorence, 2017-07-19:
HI @YuehChuan, yes let us know if you can use the kinect device with another program on your computer.

Not sure if relevant, but the error messages you are getting in the video you posted look similar to what I get when I plug my Asus Xtion into some USB 3.0 ports on some computers -- it only always works for me in a USB 2.0 port.

I'm copying @YuehChuan's error messages below to make this thread more searchable

```
No matching device found.... waiting for devices. Reason: std::string openni2_wr
apper::OpenNI2Driver::resolveDeviceURI(const string&) @ /home/schwarmcyc/LabelFu
sion/openni2-camera-lcm/src/openni2_driver.cpp @ 761 : Invalid device number 1, 
there are 0 devices connected.No matching device found.... waiting for devices. 
Reason: std::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const strin
g&) @ /home/schwarmcyc/LabelFusion/openni2-camera-lcm/src/openni2_driver.cpp @ 7
61 : Invalid device number 1, there are 0 devices connected.No matching device f
ound.... waiting for devices. Reason: std::string openni2_wrapper::OpenNI2Driver
::resolveDeviceURI(const string&) @ /home/schwarmcyc/LabelFusion/openni2-camera-
lcm/src/openni2_driver.cpp @ 761 : Invalid device number 1, there are 0 devices 
connected.No matching device found.... waiting for devices. Reason: std::string 
openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /home/schwarmc
yc/LabelFusion/openni2-camera-lcm/src/openni2_driver.cpp @ 761 :  
```

patmarion, 2017-07-21:
I spoke with a colleague who successfully used the `openini2-camera-lcm` driver with a Kinect 2.  Here's what he said was required, in addition to openni2: "i installed libfreenect2 to my system including the udev rule-- that seemed to do the trick for me"

YuehChuan, 2017-07-23:
@peteflorence good to make it searchable!
@patmarion thanks for reaching out!
Now I get kinect xbox working on docker image.
https://www.youtube.com/edit?video_id=wI-qbgAWhdU
Yeah, I think your  colleague was right. 
It is because of udev. So I think it would run nicely with kinect v2.

But I don't have kinect v2, so need to do more job on OpenNI

here are some change I made 
https://github.com/YuehChuan/LabelFusion/commit/2b64826986ca08e5d1d13448f5276937df574c44

briefly 
a.Install OpenNI driver 
b. /etc/udev rules
c. enable docker run with --privileged option 
https://github.com/YuehChuan/LabelFusion/blob/master/docker/docker_run.sh#L31

For using LabelFusion, I need to combine LCM with openni.
I search repo from openhumanoids but the driver seems not working well.  
https://github.com/openhumanoids/kinect
So I plan to use that code combine with the openni version which  mentioned above.

The way how to install openni describe inside modified compile_all.sh( thanks for giving the clue!)
https://github.com/YuehChuan/LabelFusion/blob/master/docker/compile_all.sh

New to docker and LCM,  I learnt a lot about using docker throuth this project. Thanks!

closed 2017-09-20 by YuehChuan

xxx09, 2018-08-07:
@patmarion hi, as you said:

> I spoke with a colleague who successfully used the openini2-camera-lcm driver with a Kinect 2. Here's what he said was required, in addition to openni2: "i installed libfreenect2 to my system including the udev rule-- that seemed to do the trick for me"

i can use the openini2-camera-lcm driver with Kinect 2, but how to load the kinect2's calibration [which can get from this turtorial](https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration). If i ignore the calibration data, the depth image is not aligned with rgb image and i got the following result which has large offset:
[image]
Any suggestion would be appreciated!

xxx09, 2018-08-07:
@patmarion Thank you for your reply,
The screenshot is the result of 

> run_create_data

and it is the 0000000001_color_labels.png.
I will try your advice first, and report the follow-up progress.

xxx09, 2018-08-07:
@patmarion i find when i run openni2-camera-lcm, it will record the default size (512*424) rgb and depth image, the data is too low-quality to run labelfusion. So i use rgbd_ros_to_lcm and ros_realsense pkg to replace openni2-camera-lcm. And run labelfusion successfully, thanks again for your labelfusion.

nihar0602, 2021-04-12:
Hi @xxx09,
I'm currently using intel real sense for this project and i have some doubt? 
Did you use the docker image for LabelFusion? 
If yes, then did you install this _rgbd_ros_to_lcm and ros_realsense_ in docker or was it just on your normal computer? 
Any help is appreciated. 

Thanks!