RobotLocomotion/LabelFusion #9: directorPython warning: utorso is not a valid coordinate frame

5yler, 2017-08-02:
When I try to use the `run_create_data` command (worked fine in former CorlDev) I get the following output for ever frame:

```
writing: /home/syler/spartan/src/LabelFusion/data/logs_test/donuts_realsense/images/0000000013_rgb.png
writing: /home/syler/spartan/src/LabelFusion/data/logs_test/donuts_realsense/images/0000000013_utime.txt

** (directorPython:9763): WARNING **: utorso is not a valid coordinate frame
```
Only the `utime.txt` files are saved, no images appear in the `images` folder. I'm using [this fork](https://github.com/peteflorence/director) of director, branch `pf/corl-master`. Please advise!

patmarion, 2017-08-02:
How did you get `spartan/src/LabelFusion`?  Spartan hasn't been updated for `LabelFusion` yet, it still has `spartan/src/CorlDev`.  Were you using the branch from https://github.com/RobotLocomotion/spartan/pull/119 ?  That was just a first pass and isn't ready yet.

Have you tried the docker image?  I think that is the official version we are supporting now.  That said, I think the warning about `utorso` is safe to ignore, it does not affect the program, but perhaps there is some other issue.

So even though it prints `/home/syler/spartan/src/LabelFusion/data/logs_test/donuts_realsense/images/0000000013_rgb.png` that file does not actually exist?

5yler, 2017-08-02:
Yes, I was using the `add-labelfusion` branch. My mistake, I was under the impression that was further along in development.

(And yes, despite the printout the image file does not actually exist.)

patmarion, 2017-08-02:
Could you edit the file `modules/labelfusion/imagecapture.py` and after this line:

```
        ImageCapture.writeImage(image, rgbFilename)
```

add two new lines:

```
        print image.GetDimensions()
        assert os.path.isfile(rgbFilename)
```

Then try running again?  Does it output (640, 480, 1) and then assert?

5yler, 2017-08-02:
```
Creating training images from alignment...
First extracting images...
/home/syler/spartan/src/LabelFusion/data/logs_test/realsense
logFolder =  logs_test/realsense

** (directorPython:14689): WARNING **: utorso is not a valid coordinate frame

ERROR: In /home/syler/spartan/build/director/src/vtk/Filtering/vtkAbstractImageInterpolator.cxx, line 261
vtkImageInterpolator (0xa914460): Initialize(): no image data to interpolate!

writing: /home/syler/spartan/src/LabelFusion/data/logs_test/realsense/images/0000000001_rgb.png
(0, 0, 0)
Traceback (most recent call last):
  File "/home/syler/spartan/src/LabelFusion/scripts/extractImagesFromLog.py", line 26, in <module>
    labelfusion.imagecapture.captureImages(args.logFolder)
  File "/home/syler/spartan/src/LabelFusion/modules/labelfusion/imagecapture.py", line 118, in captureImages
    ImageCapture.readFromLogFile(paths['lcmlog'], paths['images'])
  File "/home/syler/spartan/src/LabelFusion/modules/labelfusion/imagecapture.py", line 111, in readFromLogFile
    imageCapture.saveImage()
  File "/home/syler/spartan/src/LabelFusion/modules/labelfusion/imagecapture.py", line 50, in saveImage
    assert os.path.isfile(rgbFilename)
AssertionError
```

patmarion, 2017-08-02:
Hmm, I wonder if it could be an issue with the log.  Did you already run ElasticFusion on the log successfully?  Did you try visualizing the data in director?

peteflorence, 2017-08-02:
Hey @5yler, I've seen this error message before but don't remember my fix I had... I suspect it may be a path issue... I'll look into it.

peteflorence, 2017-08-02:
Syler if I am to try to recreate this issue, do you think I'll be able to do so by just checking out the `add-labelfusion` branch?

5yler, 2017-08-03:
@patmarion @peteflorence After double-checking with a different log, it seems the failure to write images was in fact due to a problem with the log. With a working log, I got output of output (640, 480, 1) and saved image files fine - though still got printouts of `** (directorPython:9763): WARNING **: utorso is not a valid coordinate frame` for each image.

peteflorence, 2017-08-03:
Thanks @5yler 

I'll leave this issue open until I purge the `utorso` warning.  It's just a warning, not error, but I'd stilll like to fix this.  Assigning myself

peteflorence, 2017-08-10:
Merry Christmas everybody!

No more of the utorso warning (pushed commit to `pf/corl-master` branch of director).

Here's how to fix your existing LabelFusion install:

- If you have docker, then inside the docker (this will terribly rebuild everything, but it works): 

```
cd
rm -rf director
./labelfusion/docker/compile_all.sh 
```

- If you have `spartan`:

```
cd spartan/director
use_spartan
makedirector
```

New LabelFusion docker installs should be fine once I push the new image to docker hub.

New spartan installs should be fine.

closed 2017-08-10 by peteflorence