RobotLocomotion/LabelFusion #18: Openni2 fails to open Xtion camera

Pozimek, 2017-11-09:
When trying to collect data from the Xtion camera via the Docker image I get the following issue:

```
**root@436e5c9570b0:/etc/udev/rules.d#** openni2-camera-lcm 
No matching device found.... waiting for devices. Reason: openni2_wrapper::OpenNI2Device::OpenNI2Device(const string&) @ /root/director-build/src/openni2-camera-lcm/src/openni2_device.cpp @ 75 : Device open failed
	Could not open "1d27/0600@1/15": Failed to open the USB device!

No matching device found.... waiting for devices. Reason: openni2_wrapper::OpenNI2Device::OpenNI2Device(const string&) @ /root/director-build/src/openni2-camera-lcm/src/openni2_device.cpp @ 75 : Device open failed
	Could not open "1d27/0600@1/15": USB device not found!

```

The same camera works fine with openni2 outside the Docker image, so it is not a problem with the camera itself. I have already tried copying over my rule files from /etc/udev/rules.d/, but it has not helped. I also cannot do a lot of debugging on my own because apt-get doesn't work inside the container, so I cannot even download usbutils to run lsusb or a text editor.

peteflorence, 2017-11-09:
Hey @Pozimek please do this:

1) close down your docker and then:
```
git fetch origin
git checkout origin/pf-add-usb-lcm
```
2) do docker_run again and see if it works?

We fixed this on that branch but haven't merged yet! 

Keeping open until we merge this PR: https://github.com/RobotLocomotion/LabelFusion/pull/15

Let me know if that works for you?

Pozimek, 2017-11-09:
It worked, thanks a lot!