RobotLocomotion/LabelFusion #20: Object mesh not showing up in Object Alignment

Pozimek, 2018-01-23:
Neither the readme nor the pipeline instructions explain how to set up the object meshes. From reading the code inside utils.py I have figured out that I need to create an object_data.yaml file inside ~labelfusion/data and that it should contain, in text form, a dictionary with info about each object. Right now the contents of my objects_data.yaml are as follows:
```
square-bottle:
  mesh: /object-meshes/square-bottle.ply
  label: square-bottle
```

Trying with the following inside objects_data.yaml also fails:
`{'square-bottle': {'mesh': '/object-meshes/square-bottle.ply', 'label': 'square-bottle'}}`

The contents of my info.yaml are:
`lcmlog: trimmed_log.lcmlog
objects: ["square-bottle"]`

The mesh itself looks fine inside Meshlab.
[image]

The reconstructed pointcloud of the scene looks fine inside Director, but the object mesh does not show up at all on the right hand side of the alightment screen when I call gr.launchObjectAlignment('square-bottle').
[image]

Any idea what has gone wrong?

Pozimek, 2018-01-23:
ADDITIONAL INFO: terminal returns this whenever I run object alignment.

```
Warning: In /build/vtk-NWL3SN/vtk-5.10.1+dfsg/IO/vtkPLYReader.cxx, line 111
vtkPLYReader (0x5e8f490): Could not open PLY file

ERROR: In /build/vtk-NWL3SN/vtk-5.10.1+dfsg/Filtering/vtkExecutive.cxx, line 756
vtkStreamingDemandDrivenPipeline (0x7004220): Algorithm vtkPLYReader(0x5e8f490) returned failure for request: vtkInformation (0x56ef9d0)
  Debug: Off
  Modified Time: 481400
  Reference Count: 1
  Registered Events: (none)
  Request: REQUEST_DATA
  FORWARD_DIRECTION: 0
  FROM_OUTPUT_PORT: 0
  ALGORITHM_AFTER_FORWARD: 1

```

Pozimek, 2018-01-23:
Resolved: turns out that I misinterpreted the source code. The mesh directory location inside object_info.yaml is not relative to labelfusion directory, but instead is an absolute path.

closed 2018-01-23 by Pozimek

peteflorence, 2018-01-23:
Hi @Pozimek, thanks for posting the issue.  This is helpful to know this wasn't clear.

If you would like to add to the Readme.md to help make this extra clear and make a Pull Request, we would gladly merge it into master!

EulerWong, 2020-03-25:
> Resolved: turns out that I misinterpreted the source code. The mesh directory location inside object_info.yaml is not relative to labelfusion directory, but instead is an absolute path.

i do not know how to make my own mesh file, do you have any idea to make mesh file?