RobotLocomotion/LabelFusion #23: Cannot find any file lcmlog-* gennerated.

hoangcuongbk80, 2018-06-25:
Hi, I am very interested in your work on LabelFusion. I am trying to create my own data with LabelFusion using Asus Xtion pro live sensor without a robot.
I installed everything through your Docker: https://hub.docker.com/r/robotlocomotion/labelfusion/.
Then, run: openni2-camera-lcm and lcm-logger as shown in the screenshots below. Also, I couldn't run cdlf && cd data/logs.
However, I couldn't see any file lcmlog-* in the directory.
Could you give me some suggestions? Thank you very much!

[image]

patmarion, 2018-06-25:
After you run the `docker_run.sh` command you will have a bash session inside the docker container.  If you run `pwd` you should see you are located at `/root`.  Your data directory will be mounted at `/root/labelfusion/data`.  `cd` to that directory.  When you create files inside that data directory then the files will be visible outside the container, too.

In the screenshots you posted, it looks like you run the `lcm-logger` command from the `/root` directory, instead of from the `/root/labelfusion/data` directory.  It does appear to work and it writes an lcm log file under `/root` inside the docker container.  But, files created at `/root` will not be visible outside the container, and will not be persistent if you exit and restart the container.  Make sure you create files under `/root/labelfusion` or `/root/labelfusion/data` if you want the files to be persistent.

patmarion, 2018-06-25:
Also, one issue in your terminal screenshots is that you attempted to use the alias `cdlf` to change directories to the labelfusion source dir, but the alias wasn't defined.  Instead, just type: `cd ~/labelfusion` and it will work better.

The missing alias is due to this issue: https://github.com/RobotLocomotion/LabelFusion/issues/24

hoangcuongbk80, 2018-06-26:
Perfect! Thank Pat for your help!

closed 2018-06-26 by hoangcuongbk80

hpf9017, 2019-09-25:
from the picture, 
it shows that @hoangcuongbk80 run the openni2-camera-lcm and lcm-logger in different docker container id.
do we need to run this two cmd in the same docker container id? @patmarion 

I try to use the docker cmd to start a new terminal with same container id,
however, in the new terminal, there is no cmd called lcm-logger,
do you know how to fix it ? thank you very much! @patmarion