RobotLocomotion/LabelFusion #25: Error when run_trim

hoangcuongbk80, 2018-06-26:
Hi, 
A number of errors occurred when running run_trim as shown in the screenshot below?
It seems the problem from  hardware driver for graphics card, and I followed these issues:
https://askubuntu.com/questions/541343/problems-with-libgl-fbconfigs-swrast-through-each-update
https://askubuntu.com/questions/834254/steam-libgl-error-no-matching-fbconfigs-or-visuals-found-libgl-error-failed-t/847980
However, they didn't help!
[image]

patmarion, 2018-06-28:
hi, im not sure what the issue is.  it is possible that the docker image is not compatible with the nvidia drivers you have installed on your host machine, or not compatible with the version of `nvidia-docker` that you have installed.  On line 1 of the dockerfile used to build the image, you can see it is based on cuda 8.0:

https://github.com/RobotLocomotion/LabelFusion/blob/master/docker/labelfusion.dockerfile#L1

maybe that line needs to be upgraded to 9.0 and rebuild the image using the `docker_build.sh` script.

I think that the error messages you are encountering are not related to any specific labelfusion tool, but would occur when using an opengl application that requires working nvidia drivers.  For example, you could try running this simple test program to verify the `nvidia-docker` works:

`glxgears`

(you may have to `apt-get` install `mesa-utils` to get the program inside the container).

hoangcuongbk80, 2018-06-29:
Hi @patmarion,
Thanks for your suggestion

hoangcuongbk80, 2018-06-29:
Hi @patmarion 
Thanks for your suggestion. The `vidia-docker` works, I tested with `glxgears`.
I also did two attempts:

-  Upgraded the line in labelfusion.dockerfile to 9.0, rebuilt the image using the docker_build.sh script, then I got a error related to VTK version, it required VTK-6.2 or greater while the Director document noticed that  "On Ubuntu the build does not require VTK to be installed. A compatible version of VTK will be downloaded (precompiled binaries) at build time."

- Reinstall ubuntu, installed cuda 8.0, but got the same error as in the screenshot above.

hoangcuongbk80, 2018-06-30:
Hi @patmarion 

- libGL error was fixed by installing nvidia-docker 1.0 instead of nvidia-docker 2.0 (latest version).

- The remaining errors: 
[image]

closed 2018-07-03 by hoangcuongbk80

yikakabu, 2018-11-27:
Hi @hoangcuongbk80 
Did you install nvidia-docker in container or host? 
Thank you for your help.

andrewqho, 2019-08-20:
How did you install nvidia-docker 1.0 over nvidia-docker 2.0? I can't seem to get the older version.