RobotLocomotion/LabelFusion #26: [run_alignment_tool]  NoneType' object has no attribute '__getitem__

hoangcuongbk80, 2018-07-02:
I run run_alignment_tool and get a error as shown in the screenshot:
[image]

peteflorence, 2018-07-02:
Hi there, you need to do run_alignment_tool from inside of a log folder.
Your current path is LabelFusion/data. The path should be something like
LabelFusion/data/logs_test/2018-07-01-01/

Also prior to running the alignment tool you’ll need to have made the 3D
reconstruction. That is done with the run_prep command.

Does that help?

hoangcuongbk80, 2018-07-03:
It works! Thank you so much!

closed 2018-07-03 by hoangcuongbk80

EthonDeep, 2018-09-22:
Hi  @peteflorence :
       i have the same question ,but it doesn't  works!
root@cd8330014561:~/labelfusion/data/logs_test/2018-09-21-01# run_alignment_tool 
Opening object alignment tool...
log folder: logs_test/2018-09-21-01
Traceback (most recent call last):
  File "/root/labelfusion/scripts/labelFusionApp.py", line 79, in <module>
    globalsDict=globals())
  File "/root/labelfusion/modules/labelfusion/setup.py", line 38, in setupLabelFusionDirector
    firstFrameToWorldTransform = utils.getFirstFrameToWorldTransform(filenames['transforms'])
TypeError: 'NoneType' object has no attribute '__getitem__'
Done.
root@cd8330014561:~/labelfusion/data/logs_test/2018-09-21-01#