RobotLocomotion/LabelFusion #27: [run_trim] Failed to get BotCamTrans for camera

hoangcuongbk80, 2018-07-03:
Hi! when run run_trim I get the following issue:: 
Failed to get BotCamTrans for camera: OPENNI_FRAME_D
Failed to get coord_frame for camera: OPENNI_FRAME_D
[image]

xxx09, 2018-08-05:
I get the same problem, have you solved it?

hoangcuongbk80, 2018-08-06:
Hi, Not yet. But, everything I need working fine without solving this problem.

yikakabu, 2018-12-04:
@hoangcuongbk80 Hi. 
When I run
> roslaunch rgbd_ros_to_lcm lcm_republisher.launch

 in catkin_ws. There is a error:
[lcm_republisher-2] process has died [pid 13851, exit code -11, cmd /home/huo/ros/catkin_ws/devel/lib/rgbd_ros_to_lcm/lcm_republisher __name:=lcm_republisher __log:=/home/huo/.ros/log/cd34ba94-f79f-11e8-862e-509a4c0a4601/lcm_republisher-2.log].
Any suggestion? Thanks.