RobotLocomotion/LabelFusion #28: How to generate file above_table_pointcloud.vtp?

hoangcuongbk80, 2018-07-04:
I am trying to segment the pointcloud above the table through run_alignment_tool, but I don't see any file like   above_table_pointcloud.vtp genneted:

[image]

5yler, 2018-07-06:
It's been a while since I've used LabelFusion but here are my notes:

- run_alignment_tool => registration
- shift+click a point on table, shift+click a point above table
F8 (Director terminal): 
- `gr.rotateAndSegmentPointcloud()`
- `gr.launchObjectAlignment('drill')`

(This was over a year ago so the process may have changed)

From your screenshot above it appears that maybe the `gr.segmentTable()` command did not get executed correctly (note that it is green in the directorPython window, as opposed to the `gr.saveAboveTablePolyData()` which is in black. Perhaps try again and first press enter in the directorPython window until you get a `>>>` at the start of your prompt?

hoangcuongbk80, 2018-07-06:
Many thanks for your help! I will check it.

hoangcuongbk80, 2018-07-09:
Resolved: Commands on directorPython window should be typed instead of copy and paste.

closed 2018-07-09 by hoangcuongbk80