RobotLocomotion/LabelFusion #29: [docker build] Dependency of  director's branch

ianre657, 2018-07-13:
I'm building my own docker image because I have upgraded the ElasticFusion to support my device.
Straight to the point, the `docker_build.sh` have some bugs and the major issue I concerned is

In file `docker/compile_all.sh` ,the function `build_director`:

```shell number
build_director()
{
  cd $root_dir
  git clone https://github.com/RobotLocomotion/director.git
  cd director
  git remote add pf https://github.com/peteflorence/director.git
  git fetch pf
  git checkout pf/corl-master
```
I can't build the director through `pf/corl-master` because of  the compatibility of VTK and Qt
however, using the `pf/master`  branch ,  it works like a charm.

Mr. @patmarion :
I'm wondering the meaning/usage of the branch `corl-master`. Is it necessary to use this branch to make LabelFusion work now ?

patmarion, 2018-07-13:
Hi Ian,

I suspect that `pf/master` is just an old version of upstream master, and `corl-master` has some modifications to `pf/master` that are required for some functions to work.  I believe those modifications have been isolated to two pull requests here:

https://github.com/RobotLocomotion/director/pull/563
https://github.com/RobotLocomotion/director/pull/564

This some minor conflict resolution those pull requests should still merge cleanly into upstream master, which is this `master` on the `robotlocomotion` org:

https://github.com/RobotLocomotion/director/tree/master

Those PRs should work fine, but I wanted some small code design changes before merging them, so that's why they got delayed and are still unmerged.  I'll see if I have some time next week to get them into master, then the docker image can just build robotlocomotion master branch.  Until then, you can do most of the `labelfusion` pipeline without those changes, but you will need those changes at the last step which goes over the lcm log and renders out the frames.  You could try applying those commits yourself, otherwise I'll let you know when updated.

In the meantime, you could post the VTK or Qt error message you got, perhaps there's an easy fix to that?

ianre657, 2018-07-17:
Hi pat,
First, I found that  `sudo` and  `lsb-release` (used in director’s superbuild ) seems not longer be included in the Official Ubuntu docker image. So we need to manually add that into `install_dependency.sh` to make things work, and the dependency of Qt5 are not automatically installed if user choose to built with that option.
 
 The following are some error messages by trying to build director( in corl-master branch )
 Because we didn’t add the ENTRYPOINT option before finish to compiling all third parties, I can’t just run the image to examine the log files.
 

1. Use the system VTK
   Ubuntu have VTK 5.10.1, not 6. It seems that Drake have update the requirement of VTK version after the completion of `pf/corl-master`’s director. I though there are two way to solve this… the first is to use an old version of Drake, and the other is upgrade the system VTK version to make these work. Giving up after spend some time trying this, can't figure out where the 

```
    CMake Error at cmake/externals.cmake:334 (message):
      Director requires VTK version 6.2 or greater.  System has VTK version
      5.10.1
```
  
2. -DUSE_SYSTEM_VTK:BOOL=OFF

 Failed in building PCL1.8, no further message given.
```
 In file included from /root/director-build/src/pcl/sample_consensus/include/pcl/sample_consensus/sac_model.h:52:0,
                     from /root/director-build/src/pcl/sample_consensus/include/pcl/sample_consensus/sac.h:45,
                     from /root/director-build/src/pcl/sample_consensus/include/pcl/sample_consensus/ransac.h:44,
                     from /root/director-build/src/pcl/registration/include/pcl/registration/icp.h:45,
                     from /root/director-build/src/pcl/recognition/src/face_detection/rf_face_detector_trainer.cpp:17:
    /root/director-build/src/pcl/sample_consensus/include/pcl/sample_consensus/model_types.h: In function 'void __static_initialization_and_destruction_0(int, int)':
    /root/director-build/src/pcl/sample_consensus/include/pcl/sample_consensus/model_types.h:99:3: warning: 'pcl::SAC_SAMPLE_SIZE' is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
       SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
       ^
    /root/director-build/src/pcl/sample_consensus/include/pcl/sample_consensus/model_types.h:99:3: note: declared here
    
    ...(ignore)
    
    -- Installing: /root/install/include/pcl-1.8/pcl/stereo/impl/disparity_map_converter.hpp
    [ 79%] Completed 'pcl'
    [ 79%] Built target pcl
    Makefile:83: recipe for target 'all' failed
```

3. USE_PRECOMPILED_VTK: ON
  (VTK7.1 + QT5.5)
  - Qt5: (-DDD_QT_VERSION:STRING=4)
```
  failed while building QtPropertyBrowser
    [ 36%] Completed 'QtPropertyBrowser'
    [ 36%] Built target QtPropertyBrowser
    Makefile:83: recipe for target 'all' failed
    make: *** [all] Error 2
```  
- Qt4:
  same as `-DUSE_SYSTEM_VTK:BOOL=OFF`
  
4. Use precompiled VTK:OFF
  - -DDD_QT_VERSION:STRING=4
    (Vtk8+Qt4)
```  
failed while building `vtk` ,  no further message given.
    -- Set runtime path of "/root/install/bin/vtkpython" to ""
    [ 73%] Completed 'vtk'
    [ 73%] Built target vtk
    make: *** [all] Error 2
    Makefile:83: recipe for target 'all' failed
    The command '/bin/sh -c /tmp/compile_all.sh' returned a non-zero code: 2
   ```
  
  - -DDD_QT_VERSION:STRING=5
    (Vtk8+ Qt5)
  failed while building `flann` ,  no further message given.
  ```
  -- Installing: /root/install/include/flann/nn/ground_truth.h
    -- Installing: /root/install/bin/flann_example_cpp
    -- Set runtime path of "/root/install/bin/flann_example_cpp" to ""
    -- Installing: /root/install/share/doc/flann/manual.pdf
    [ 36%] Completed 'flann'
    [ 36%] Built target flann
    Makefile:83: recipe for target 'all' failed
    make: *** [all] Error 2
```

While I’m trying  another option( build my own VTK), I found https://github.com/RobotLocomotion/director/pull/508, and start to suspect that corl-master is a branch not maintained anymore, After built successfully with the master branch, I opend this issue.

ianre657, 2018-07-17:
The following are some error messages by trying to build director( in corl-master branch )
 Because we didn’t add the ENTRYPOINT option before finish to compiling all third parties, I can’t just run the image to examine the log files.
 
1. Use the system VTK
   Ubuntu have VTK 5.10.1, not 6. It seems that Drake have update the requirement of VTK version after the completion of `pf/corl-master`’s director. I though there are two way to solve this… the first is to use an old version of Drake, and the other is upgrade the system VTK version to make these work. Giving up after spend some time trying this, can't figure out where the 

```
    CMake Error at cmake/externals.cmake:334 (message):
      Director requires VTK version 6.2 or greater.  System has VTK version
      5.10.1
```
  
2. -DUSE_SYSTEM_VTK:BOOL=OFF

 Failed in building PCL1.8, no further message given.
```
 In file included from /root/director-build/src/pcl/sample_consensus/include/pcl/sample_consensus/sac_model.h:52:0,
                     from /root/director-build/src/pcl/sample_consensus/include/pcl/sample_consensus/sac.h:45,
                     from /root/director-build/src/pcl/sample_consensus/include/pcl/sample_consensus/ransac.h:44,
                     from /root/director-build/src/pcl/registration/include/pcl/registration/icp.h:45,
                     from /root/director-build/src/pcl/recognition/src/face_detection/rf_face_detector_trainer.cpp:17:
    /root/director-build/src/pcl/sample_consensus/include/pcl/sample_consensus/model_types.h: In function 'void __static_initialization_and_destruction_0(int, int)':
    /root/director-build/src/pcl/sample_consensus/include/pcl/sample_consensus/model_types.h:99:3: warning: 'pcl::SAC_SAMPLE_SIZE' is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
       SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
       ^
    /root/director-build/src/pcl/sample_consensus/include/pcl/sample_consensus/model_types.h:99:3: note: declared here
    
    ...(ignore)
    
    -- Installing: /root/install/include/pcl-1.8/pcl/stereo/impl/disparity_map_converter.hpp
    [ 79%] Completed 'pcl'
    [ 79%] Built target pcl
    Makefile:83: recipe for target 'all' failed
```

3. USE_PRECOMPILED_VTK: ON
  (VTK7.1 + QT5.5)
  - Qt5: (-DDD_QT_VERSION:STRING=4)
```
  failed while building QtPropertyBrowser
    [ 36%] Completed 'QtPropertyBrowser'
    [ 36%] Built target QtPropertyBrowser
    Makefile:83: recipe for target 'all' failed
    make: *** [all] Error 2
```  
- Qt4:
  same as `-DUSE_SYSTEM_VTK:BOOL=OFF`
  
4. Use precompiled VTK:OFF
  - -DDD_QT_VERSION:STRING=4
    (Vtk8+Qt4)
```  
failed while building `vtk` ,  no further message given.
    -- Set runtime path of "/root/install/bin/vtkpython" to ""
    [ 73%] Completed 'vtk'
    [ 73%] Built target vtk
    make: *** [all] Error 2
    Makefile:83: recipe for target 'all' failed
    The command '/bin/sh -c /tmp/compile_all.sh' returned a non-zero code: 2
   ```
  
  - -DDD_QT_VERSION:STRING=5
    (Vtk8+ Qt5)
  failed while building `flann` ,  no further message given.
  ```
  -- Installing: /root/install/include/flann/nn/ground_truth.h
    -- Installing: /root/install/bin/flann_example_cpp
    -- Set runtime path of "/root/install/bin/flann_example_cpp" to ""
    -- Installing: /root/install/share/doc/flann/manual.pdf
    [ 36%] Completed 'flann'
    [ 36%] Built target flann
    Makefile:83: recipe for target 'all' failed
    make: *** [all] Error 2
```

While I’m trying  another option( build my own VTK), I found https://github.com/RobotLocomotion/director/pull/508, and start to suspect that corl-master is a branch not maintained anymore, After built successfully with the master branch, I opened this issue.

ianre657, 2018-10-06:
Hi @patmarion,
I have figured this out. To build Labelfusion, an older version of Director is needed. Also, It looks like that version of Director is not compatible with VTK7, so I use VTK5 instead.

I have write a newer version of the Dockerfile which can actually work and, support  `nvidia-docker2`!
Link of the project is here:
https://github.com/ianre657/LabelFusion

The docker image is on the dockerhub. If you are interested, it's at: 
 ianre657/labelfusion:16.04-latetest

We are still trying to fix some bugs, and will make a PR once we have finished this.

peteflorence, 2018-10-14:
Hi @ianre657, that seems pretty useful you have worked through some of these compatibility issues.  Would be happy to look at a PR once you have finished!

usama0300854, 2025-11-08:
@ianchen-tw  can you help me setup labelfusion docker image! please