RobotLocomotion/LabelFusion #31: Error response from daemon while installing LabelFusion

PranjaLBiswas27, 2018-08-21:
Hi, 

I was trying to install LabelFusion via docker, but on running the following command:

LabelFusion/docker/docker_run.sh /path/to/data-folder

I get the following error after bash finishes pulling the image:

docker: Error response from daemon: create LabelFusion/: "LabelFusion/" includes invalid characters for a local volume name, only "[a-zA-Z0-9][a-zA-Z0-9_.-]" are allowed. If you intended to pass a host directory, use absolute path.
See 'docker run --help'.
non-network local connections being removed from access control list

Before I had already installed docker as instructed in [here](https://github.com/NVIDIA/nvidia-docker). I already have Cuda 8 and cudnn 6. Please let me know what is the issue. Thanks a lot in advance.

Kind Regards
Pranjal Biswas

peteflorence, 2018-10-14:
Hi @PranjaLBiswas27, sorry I didn't see this before, but happy to help you work through this if you're still interested.  To clarify, `/path/to/data-folder` should be something like `/media/peteflorence/labelfusion-data` for your computer, not sure if that's helpful.

PranjaLBiswas27, 2019-01-22:
Sorry but I abandoned work on this so I dont know if your solution works or not. I will just close this thread for now.

closed 2019-01-22 by PranjaLBiswas27