RobotLocomotion/LabelFusion #33: Error when generating labeled images

YLP95, 2018-09-18:
When I execute "run_create_data", the images from LCM log has been successfully extracted. But generating labeled images is failured.
[image]

closed 2018-09-18 by YLP95

patmarion, 2018-09-18:
Hi, did you find a solution?  I just looked at the code, and it looks like in this case the `self.colors` variable is hardcoded to be:

```
array([[0.        , 0.        , 0.        , 1.        ],
       [0.52024118, 0.        , 0.58692157, 1.        ],
       [0.        , 0.        , 0.78826863, 1.        ],
       [0.        , 0.52158824, 0.8667    , 1.        ],
       [0.        , 0.6667    , 0.62746471, 1.        ],
       [0.        , 0.61306863, 0.        , 1.        ],
       [0.        , 0.85362157, 0.        , 1.        ],
       [0.51762353, 1.        , 0.        , 1.        ],
       [0.97253529, 0.85488824, 0.        , 1.        ],
       [1.        , 0.36470588, 0.        , 1.        ],
       [0.85362157, 0.        , 0.        , 1.        ],
       [0.8       , 0.8       , 0.8       , 1.        ],
       [0.5       , 0.5       , 0.5       , 1.        ]])
```

It will be indexed with the label value read from `object_data.yaml`.  In our dataset, this yaml file contains:

```
oil_bottle:
  label: 1
  mesh: object-meshes/handheld-scanner/oil_bottle_small.vtp
phone:
  label: 2
  mesh: object-meshes/phone_handheld.vtp
robot:
  label: 3
  mesh: object-meshes/robot_handheld.vtp
toothpaste:
  label: 4
  mesh: object-meshes/toothpaste_aligned.vtp
tissue_box:
  label: 5
  mesh: object-meshes/tissue_box_subdivision_colors.vtp
blue_funnel:
  label: 6
  mesh: object-meshes/blue_funnel.stl
drill:
  label: 7
  mesh: object-meshes/handheld-scanner/drill_decimated_mesh.vtp
cracker_box:
  label: 8
  mesh: object-meshes/ycb_models/003_cracker_box/textured.obj
tomato_soup:
  label: 9
  mesh: object-meshes/ycb_models/005_tomato_soup_can/textured.obj
spam:
  label: 10
  mesh: object-meshes/ycb_models/010_potted_meat_can/textured.obj
mug:
  label: 11
  mesh: object-meshes/ycb_models/025_mug/textured.obj
bowl:
  label: 12
  mesh: object-meshes/ycb_models/024_bowl/textured.obj
```

So the `label` value should be an integer between 0 and 12.  Is it possible you have a `object_data.yaml` file which contains non integer values in the place of the `label` field?

YLP95, 2018-09-18:
> Hi, did you find a solution? I just looked at the code, and it looks like in this case the `self.colors` variable is hardcoded to be:
>
> ```
> array([[0.        , 0.        , 0.        , 1. [...]

Thank you very much.
The problem has been solved. The reason is the label value in object_data.yaml is wrong.