RobotLocomotion/LabelFusion #36: labels are a bit off

hoangcuongbk80, 2018-10-23:
hi,

I am trying to replicate your samples, but label images generated are a bit off as shown in the image.
Any suggestion?

[image]

MengDes, 2019-04-27:
> Hi, Thanks for your reply.
> 
> I Played around and tried to have a better registration result. As you can see in the image below, it looks good. But, the label images are still a bit off, especially on top of the object.
> 
> [image]
> [image]
> [image]

have you solve this problem? i have drag it around to make it look good ,but in generated image the result still a bit of , and i have setting intrinsic

tiexuedanxin, 2020-06-08:
have you solve this problem? I come across the same problem, could you give me some advice on how to solve the problem, thank you for advance