RobotLocomotion/LabelFusion #38: Use labelfusion out of docker

CreateBetter, 2018-10-26:
Hello~Thanks so much for your labelfusion, It helps me a lot.But now I meet some question,I want to use Intel D415 camera in labelfusion docker. But I can not drive the camera in the docker. Although I have already used ORBBEC Astra pro with rgbd_ros_to_lcm[https://github.com/MobileManipulation/rgbd_ros_to_lcm](url).
So I want to ask if I can use labelfusion out of docker, how to pull away labelfusion from docker?
Thanks again~

ianre657, 2018-11-28:
hey @CreateBetter ,
I just run Labelfusion with intel D435 successfully (which comes with nvidia-docker2)
maybe you can try my forked one. 
https://github.com/ianre657/LabelFusion

chaiban, 2019-08-05:
Hey @ianre657 ,
Thank you for the forked version for D435. I am trying to use labelfusion for my own dataset using Realsense D435. I followed your instructions but there is nothing in log files. Generated lcmlog file does not contain any information. How do I fix it ?

Thank you.

hpf9017, 2019-10-12:
@chaiban
https://github.com/RobotLocomotion/LabelFusion/issues/27#issuecomment-411358713
see this comment, it may help.