RobotLocomotion/LabelFusion #39: Missing depth images

vevenom, 2018-11-16:
Hi,

i like what you did with LabelFusion pipeline. Very nice!

I wanted to conduct experiments on your included dataset but I noticed that raw depth images are missing from the main dataset. However, for the demo sample dataset they are included. Was this intentional? Is there any way of recovering the original depth maps from the dataset?

Thanks for the answer in advance.

patmarion, 2018-11-19:
You can extract and save the depth images from the data you have already downloaded.  The script you can use is named `extractImagesFromLog.py`.

For example:

```
# a log folder is a folder that contains "original_log.lcmlog"
# such as ~/labelfusion/data/logs_test/2017-06-16-20
cd <path to log-folder>

# use the --saveDepth=True option
directorPython $LABELFUSION_SOURCE_DIR/scripts/extractImagesFromLog.py --saveDepth=True --logFolder $PWD --bot-config $LABELFUSION_SOURCE_DIR/config/bot_frames.cfg
```

You'll have to run this for each log.  Sorry for the inconvenience!  In the future we may re-upload the data organized by log folder, and will include depth at that time.

vevenom, 2018-11-20:
Good, thank you for your help :)

closed 2018-11-20 by vevenom