RobotLocomotion/LabelFusion #41: About the format of poses.yaml

yikakabu, 2018-11-28:
Hi , Thank you for your awesome work.
I am confused on the output of "run_create_data." 
Here are the logs folder I got. 
```
logs/
        2018-11-28-00/
                info.yaml
                original_log.lcmlog
                images/
                        ..._rgb.png
                        ..._color_labels.png
                        ..._labels.png
                        ..._utime.txt
                        ..._poses.yaml
                above_table_pointcloud.vtp
                reconstructed_pointcloud.vtp
                converted_to_ascii.ply
                converted_to_ascii_modified_header.ply
                posegraph.posegraph
                registration_result.yaml
                transforms.yaml
```
1. There is a 'pose' array like 'pose: [[0.14957513121265414, -0.02489842391691366, 1.277904559798256], [-0.49159512738187244, 0.5768455565004487, 0.6321354755944216, -0.1612084835705075]]' .
Could you please explain what those number present for ?
2. *labels.png just looks like image that pure black. How can I understand those *labels.png file
[image]
Thank you for your help !

patmarion, 2018-11-30:
Hi,

- The pose array is: `[position, quaternion]` as `[[x,y,z], [w, x, y, z]]`

- The `*_labels.png` image is a single channel image that contains values in the range of [0, max_label].  When viewed with a standard image viewer, it displays these values to colors, but you'll just see black and dark gray shades that are indistinguishable from black.  The `*_color_labels.png` image is provided for convenience, it displays the labels mapped with colors from blue to red.  You can view the `*_labels.png` image with this script to see the labels and a histogram of labels:

https://github.com/RobotLocomotion/LabelFusion/blob/master/scripts/plotLabels.py

Just run `python plotLabels.py <image-file>`

yikakabu, 2018-12-05:
@patmarion 
Thanks a lot. I got it.
There is a problem when I run my own data collected by Kinect. The scene reconstructed by "run_prep" are much smaller than the model I imported. 
So, is there any process to make sure the size of model and scene match?
Thanks for your reply.

closed 2019-03-22 by yikakabu