RobotLocomotion/LabelFusion #44: About the 3D bounding box

fjmomy, 2018-12-10:
How can I get the picture with 3D bounding box? 
If it can be got by your code directly? What command should i run?
Thank you！

yikakabu, 2018-12-10:
@fjmomy 
I have the same problem. The sample data only have one label image contains 3d bbox. And the images I created by LabelFusion don't contain bbox.
Let me know if you have any ideas.Thank U.

patmarion, 2018-12-10:
Hi,

The code to render the bounding box was part of some work in progress code that I wrote to make a video.  I have put that code into a standalone PR here:

https://github.com/RobotLocomotion/LabelFusion/pull/45

You can apply that diff locally to enable the bounding  box rendering.  Please let me know if it works, then I will land the pull request.  Thanks!

fjmomy, 2018-12-11:
Following your suggestion, I changed the code on my computer and got the pics with 3D bbox.
Thanks for your work, it really help me a lot!

F-jie, 2018-12-17:
Hi, I wonder how do you visualize the reconstructed scenery？VTK or something else relate to it？
Thanks！
@patmarion