RobotLocomotion/LabelFusion #46: How to get pointcloud from dataset?

moningkai, 2019-01-15:
Hello every one!
1.I already know the boundry of an object in one frame image,how can I get the point cloud of this object in this frame?
2.if I have point cloud of this frame, I need rgb-camera intrinsic matrix and rgb-camera-to-depth-camera transform matrix.So I can get the mapping from point cloud to rgb-image.Where could I obtain these matrixs? 
Thanks for your rely.

patmarion, 2019-01-16:
Hi!  Sorry for the slow response.  I'll give some brief answers now and maybe we can elaborate more in the comments.

1. do you mean the pointcloud derived from the depth image frame, or the pointcloud of the scene reconstruction?

2. In LabelFusion the camera intrinsics are stored in this file, under the name `OPENNI_FRAME_LEFT` in the `cameras` group:

https://github.com/RobotLocomotion/LabelFusion/blob/master/config/bot_frames.cfg

Note that these are unique to your camera.  The file distributed with LabelFusion contains default values.

moningkai, 2019-01-20:
@patmarion thank you very much! now I still have two questions.
1.Does the object have a same position in depth image and rgb image?
2.How to get pointcloud from depth image?  
I want to get the pointcloud of a object from depth image