RobotLocomotion/LabelFusion #47: On executing ElasticFusion to get the posegraph and ply I get a core dump moreover docker_build.sh also fails

sunit1409, 2019-02-13:
root@3c0556edc37a:~/ElasticFusion/GUI/build# ./ElasticFusion -l /root/labelfusion/data/trimmed_log.lcmlog -f
rewind is: 0
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X11 Error: BadMatch (invalid parameter attributes)
X11 Error: BadValue (integer parameter out of range for operation)
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Failed to create an OpenGL context
Aborted (core dumped)

on doing docker build I get the following error.

CMake Error at cmake/externals.cmake:334 (message):
  Director requires VTK version 6.2 or greater.  System has VTK version
  5.10.1
Call Stack (most recent call first):
  CMakeLists.txt:27 (include)

-- Configuring incomplete, errors occurred!
See also "/root/director-build/CMakeFiles/CMakeOutput.log".
See also "/root/director-build/CMakeFiles/CMakeError.log".
The command '/bin/sh -c /tmp/compile_all.sh' returned a non-zero code: 1

peteflorence, 2019-02-14:
Hi @sunit1409, to start out, I would recommend not building the docker image, but just getting our provided docker image from dockerhub: https://hub.docker.com/r/robotlocomotion/labelfusion/

sunit1409, 2019-02-14:
Yes I am using the docker image and in that I get the following error.
root@3c0556edc37a:~/ElasticFusion/GUI/build# ./ElasticFusion -l /root/labelfusion/data/trimmed_log.lcmlog -f
rewind is: 0
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X11 Error: BadMatch (invalid parameter attributes)
X11 Error: BadValue (integer parameter out of range for operation)
terminate called after throwing an instance of 'std::runtime_error'
what(): Pangolin X11: Failed to create an OpenGL context
Aborted (core dumped)

peteflorence, 2019-02-14:
Did you install nvidia-docker1, or nvidia-docker2?  It seems like you're
getting OpenGL issues.  nvidia-docker2 needs a couple tricks to get OpenGL
to work, we haven't added them to the instructions yet, but there is lots
of guidance on the web on how to do this.  nvidia-docker1 should work fine.
Let us know!

sunit1409, 2019-02-15:
I had installed nvidia docker-2 as was mentioned in @ianre657 's code , well I will check out the open GL issues and update on this thread.