RobotLocomotion/LabelFusion #48: elastic fusion issues with lcmlog

sunit1409, 2019-02-14:
I am trying to install the software from scratch but when I pass the lcmlog to the elastic fusion as required in run_prep it throws a seg fault. elastic fusion by defaults looks for .klg files. Is it because of that, is there any modification in the elastic fusion that you are using.

manuelli, 2019-02-14:
Hi Sunit,

Are you using the software via docker? If so we use a custom branch of ElasticFusion that has an LCM reader. It is built as part of the docker image, see [here](https://github.com/peteflorence/ElasticFusion.git). The branch is [this](one). It should read `.lcm` files. You might just want to check inside your docker image and make sure that ElasticFusion is pointing the correct branch.

sunit1409, 2019-02-14:
No not the docker inage. On using the docker image I get the following error. 
root@3c0556edc37a:~/ElasticFusion/GUI/build# ./ElasticFusion -l /root/labelfusion/data/trimmed_log.lcmlog -f
rewind is: 0
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X11 Error: BadMatch (invalid parameter attributes)
X11 Error: BadValue (integer parameter out of range for operation)
terminate called after throwing an instance of 'std::runtime_error'
what(): Pangolin X11: Failed to create an OpenGL context
Aborted (core dumped)

on doing docker build I get the following error.

CMake Error at cmake/externals.cmake:334 (message):
Director requires VTK version 6.2 or greater. System has VTK version
5.10.1
Call Stack (most recent call first):
CMakeLists.txt:27 (include)

-- Configuring incomplete, errors occurred!
See also "/root/director-build/CMakeFiles/CMakeOutput.log".
See also "/root/director-build/CMakeFiles/CMakeError.log".
The command '/bin/sh -c /tmp/compile_all.sh' returned a non-zero code: 1

manuelli, 2019-02-14:
Ok so our software is really only designed to work inside the docker image. That being said if you want to get it working without docker looking at the dockerfile is a good place to start. 

I will look into the `docker_build` issues you are having. Those should be fixed by updating VTK inside the container. With regards tot eh OpenGL context are you running with `nvidia-docker`? You will need an Nvidia GPU.

sunit1409, 2019-02-15:
Actually I had installed nvidia-docker 2 in my system and I am using nvidia 1080Ti.

khmariem, 2019-09-24:
@sunit1409 I think it will work if you downgrade to nvidia-docker 1.

ianre657, 2019-12-08:
@sunit1409  

This version of labelfusion have been upgraded to nvidia-docker2
https://github.com/RobotLocomotion/LabelFusion/pull/40

The docker image is provided here, you might want to check it out:
https://hub.docker.com/repository/docker/ianre657/labelfusion

zx19950617, 2019-12-15:
@ianre657 Hi, ian! Thanks a lot for your contributions to the upgraded labelfusion. I used your version of labelfusion, but encountered one problem when running the command "run_prep", shown as follows:

roor@3ca7af96544d:~/labelfusion/data/logs/sampleTest# run_prep
Running pre-processing stage...
rewind is: 0
error: XDG_RUNTIME_DIR not set in the environment.

Then, no any GUI. Could you give me some advice about how to solve above problem, please?
Once again, thanks a lot for your any help.

ianre657, 2019-12-15:
@zx19950617 

These might be the things you're looking for

+ https://github.com/autowarefoundation/autoware/issues/612
+ https://wiki.archlinux.org/index.php/Running_GUI_applications_as_root#Xhost

MrLuer, 2020-01-09:
> @sunit1409
> 
> This version of labelfusion have been upgraded to nvidia-docker2
> #40
> 
> The docker image is provided here, you might want to check it out:
> https://hub.docker.com/repository/docker/ianre657/labelfusion

hi, thank for your branch of realsense https://github.com/ianre657/LabelFusion
I fellow the  README instruction，but failed to record data from realsense D435， wether it normal there is no window to be created when record data？I thought what the camera captured should be displayed，but it not happened；the lcmlog* file sometimes always 0 KB sometimes  not. so I have no idea what to do

xerifg, 2021-05-08:
> @ianre657 Hi, ian! Thanks a lot for your contributions to the upgraded labelfusion. I used your version of labelfusion, but encountered one problem when running the command "run_prep", shown as [...]

l also had the same problem.could you tell me how you solved it?thank you so much