RobotLocomotion/LabelFusion #50: The object model doesn't align to the point cloud.

yikakabu, 2019-04-25:
Hi.
In the `camera alignment tool`, I have picked the three points to align the model and it seems completed well as the first picture.
[image]
Simultaneously, as the second picture, the affordance in the `Director` is wrong and different from the `camera alignment tool`.
[image]
How should I fix this.
Thank you.

MengDes, 2019-04-27:
i think your question raised by the Jar‘s 3d model
in the second picture ，only left part points turn to green，this means you only use these points to icp with model. open run_alignment_tool, in the affordances , change the **surface mode** from **surface** to **points**, you can see only top and bottom have points. side surface empty
i meat this question too and i think you should find 3d model with points on side surface，in another way，you can adjust the icp result by hand  as mentiond in the repo'guide. 
hope this help

hpf9017, 2019-10-12:
@MengDes , how to adjust the icp result by hand?

According to the repo's guide,

If the results are inaccurate, you can rerun the above command, or you can double-click on each affordance and move it with an interactive marker: left-click to translate along an axis, right-click to rotate along an axis.

but I don't quite understand that operation, what does the affordance mean,
I double click the three points, but the interactive marker doesn't appear.