RobotLocomotion/LabelFusion #51: How to visualize color object model instead of gray one?

hoangcuongbk80, 2019-05-14:
Hi,

Could I display object model in the step "Align each object and crop point clouds" with color, my mesh original model has color. but it become gray when load it for alignment.
Gray:
[image]

color:
[image]

patmarion, 2019-05-16:
You can right click the object and change its color properties in the context menu, or select the object and find its properties in the Properties Panel.  Look at the Color By property.  Do you see the RGB colors listed as an option?  Are the colors per-vertex rgb colors, or is it a textured mesh (with color coming from an image file)?

hoangcuongbk80, 2019-05-22:
You are right! Thank youuu!!!

closed 2019-05-22 by hoangcuongbk80