RobotLocomotion/LabelFusion #52: Manual alignment - no interactive marker

edith-langer, 2019-05-21:
Hi,

I used your test data to get familiar with LabelFusion. I tried to align the drill and wanted to manually adjust the alignment. I tried double clicking on the red spheres in the scene and also on the object mesh itself. I tried the same also in the second window showing the object. But no interactive marker is appearing. 
Any ideas what I am doing wrong?

Thanks,
Edith

manuelli, 2019-05-21:
Hi Edith,

To get the interactive marker to appear you need to click on the object mesh in the main window that shows the full 3D reconstruction. Sometimes it can be hard for the UI to detect you are clicking on the object (rather than the pointcloud of the reconstruction). So you can try toggling the reconstruction off (using the radio buttons on the left hand side of the UI) and then double clicking the mesh. Then you can toggle the pointcloud back on. Hopefully that helps

edith-langer, 2019-05-22:
Thank you! I tried to get the markers in the Alignment Tool window...
It works nicely in the main window :)

closed 2019-06-25 by edith-langer