RobotLocomotion/LabelFusion #54: [run_prep] ElasticFusion error

Pai-Sho, 2019-06-26:
Hello,

I am attempting to create my own dataset using the provided docker image and an ASUS Xtion PRO camera. Everything up to the run_prep stage seems to be fine, then I get this error:

```
root@5cc4d322735d:~/labelfusion/data/logs_test/2018-06-26-01# run_prep
Running pre-processing stage...
rewind is: 0
calling rewind
LcmLogReader::rewind
timestamp: 1561575058500851
frame: 0
before jpeg read data
imageSize: 94909
about to read jpeg_read_scanlines
finished read jpeg_read_scanlines
finished jpeg_finish_decompress
finished jpeg_destroy_decompress
after jpeg read data
Error: invalid texture reference: /root/ElasticFusion/Core/src/Cuda/convenience.cuh:68
mv: cannot stat 'trimmed_log.lcmlog.posegraph': No such file or directory
sh: 1: cannot open ./trimmed_log.lcmlog.ply: No such file
Traceback (most recent call last):
  File "/root/labelfusion/scripts/prepareForObjectAlignment.py", line 63, in <module>
    with open("./converted_to_ascii.ply") as infile:
IOError: [Errno 2] No such file or directory: './converted_to_ascii.ply'
Done.
```

My host Machine is running 18.04LTS, CUDA 10.2 with a 1080Ti driving the monitor and a headless 2080Ti I use for training. I am running nvidia-docker 1.0 because I was getting a bunch of OpenGL errors with nvidia-docker2. My nvidia-smi within the docker looks like this:

```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 430.26       Driver Version: 430.26       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce RTX 208...  Off  | 00000000:01:00.0 Off |                  N/A |
| 41%   45C    P0    66W / 260W |     11MiB / 11019MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   1  GeForce GTX 108...  Off  | 00000000:02:00.0  On |                  N/A |
|  0%   39C    P2    55W / 250W |   1452MiB / 11175MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
+-----------------------------------------------------------------------------+
```

hygxy, 2019-07-20:
@Pai-Sho hey, did you solve it by now? same problem here.

khmariem, 2019-09-24:
Can you run `nvcc -V` inside your container and make sure that the compilation tools are of version 8? Also make sure that you really have nvidia-docker-1.0 by looking at `nvidia-docker version`. I overcame that error by using nvidia-docker-1.0, so that was the issue as far as I remember. Good luck !

huzheyuan740, 2019-10-25:
@hygxy Have you solved the problem? I have the same problem here. Thank you in advance.

MrLuer, 2020-01-22:
hey,I have the same problem, do you have any solution?

TommasoBendinelli, 2020-01-23:
Same issue here

JannisEgelhof, 2020-02-13:
maybe this helps:
https://github.com/mp3guy/ElasticFusion/issues/170 #

saprrow, 2021-04-01:
Same issue here

luojianbin2006, 2021-11-04:
Same issue. My solution:
my GPU is Quadro RXT5000 which CUDA Capability is 7.5. So I add 75 in line of setting CUDA_ARCH_BIN of the Cmakefile.txt in ~/ElasticFuion/Core/src, and install cuda10. Then remake ElasticFuion.

WinglokWong616, 2022-01-04:
Same issue here when running **ianchen-tw/LabelFusion** with Nvidia-Docker2. Did someone make some progress on this probelm yet? What's the cause?

WinglokWong616, 2022-01-05:
> Same issue. My solution: my GPU is Quadro RXT5000 which CUDA Capability is 7.5. So I add 75 in line of setting CUDA_ARCH_BIN of the Cmakefile.txt in ~/ElasticFuion/Core/src, and install cuda10. Then remake ElasticFuion.

This did help, Thanks! I just changed the Cmakefile and remake, then everything worked out.