RobotLocomotion/LabelFusion #57: TypeError: 'NoneType' object has no attribute '__getitem__'

zx19950617, 2019-07-10:
Hi, everybody!I run the alignment tool, and then the error is shown as follow. What should I check?Thanks for any suggestion.

root@ed246d79bb25:~/labelfusion/data/LabelFusion_Sample_Data/logs/2017-06-13-29#
 run_alignment_tool 
Opening object alignment tool...
log folder: logs/2017-06-13-29
Traceback (most recent call last):
  File "/root/labelfusion/scripts/labelFusionApp.py", line 79, in <module>
    globalsDict=globals())
  File "/root/labelfusion/modules/labelfusion/setup.py", line 38, in setupLabelFusionDirector
    firstFrameToWorldTransform = utils.getFirstFrameToWorldTransform(filenames['transforms'])
TypeError: 'NoneType' object has no attribute '__getitem__'
Done.

xyli001, 2019-07-30:
I have met this problem too. At the beginning, all the system is OK,after I had installed the ROS, and run nvidia-docker again, the problem accured. 
"
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast

andrewqho, 2019-08-09:
DId you manage to figure this out? I am running into a similar issue. I am running Ubuntu 18.04.2 LTS.

sunmaize, 2019-08-13:
I have the libGL issue too.
root@9c9bcc370987:~/labelfusion/data/logs/2017-06-13-29# run_alignment_tool
Opening object alignment tool...
log folder: logs/2017-06-13-29
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
using user specified transform
loading above table pointcloud

I can see the GUI,but i can't see the pointcloud.

patmarion, 2019-08-14:
Hi, the error message posted in the origin comment:

```
  File "/root/labelfusion/modules/labelfusion/setup.py", line 38, in setupLabelFusionDirector
    firstFrameToWorldTransform = utils.getFirstFrameToWorldTransform(filenames['transforms'])
TypeError: 'NoneType' object has no attribute '__getitem__'
```

occurs because the variable `filenames` is `None`, which could happen because the tool is unable to find the file `info.yaml` in the requested log directory.  Please double check that you launched the tool while in the correct working directory and have the correct log folder argument.  When the tool launches it should print the location of the log folder is it using:

```
print 'log folder:', args.logFolder
```

Sorry that the error message is not more descriptive.  You could edit the function `setupLabelFusionDirector` to do better error checking and print a more helpful message, then pull request it!

closed 2019-12-04 by zx19950617

chrisxu1995, 2021-01-27:
Hi, I have the same issue, did you fixed it?

zx19950617, 2021-01-27:
@chrisxu1995 Please check the /path/to/data-folder in command "LabelFusion/docker/docker_run.sh /path/to/data-folder" to make sure it  covers LabelFusion_Sample_Data folder. I fixed the problem just through this.

iamlucaswolf, 2021-06-09:
Adding to @zx19950617 answer, also note that docker expects _absolute_ paths for mounting volumes (which is what docker_run.sh does). However, you can do something like

```bash
$ LabelFusion/docker/docker_run.sh "$(pwd)/relative/path/to/data"
```

Hope that helps! :)

Mechazo11, 2024-09-09:
@iamlucaswolf thank you for adding on answer to @zx19950617 . I know this probably three years late but the official tutorial provides

```bash
cd ~/labelfusion/data/logs_test/2017-06-16-20
run_alignment_tool
```  
after running the docker image. So doesn't that mean we need to copy the sample dataset inside the docker container beforehand? I did try it but am getting ```NoneType``` error.

Mechazo11, 2024-09-09:
For folks who are using this tool after this many years, to avoid getting this error, copy the contents of the sample dataset into the docker container using the following steps

* Run the docker image
* Find container ID ```docker ps```
* Download and unzip the sample dataset
* Copy **logs**, **object-meshes**, **object_data.yaml** directories and file respectively one at a time using ```docker cp``` command. An example is given below. Note, I downloaded the sample dataset and unzipped it in ```Downloads``` folder.

```bash
cd ~/Downloads
docker cp LabelFusion_Sample_Data/logs f262fcc7fc6c:/root/labelfusion/data
```

where ```f262fcc7fc6c``` is the container ID obtained in the previous step.