RobotLocomotion/LabelFusion #59: bad alignment results

hygxy, 2019-07-29:
Hey, I was using this tool to create my own data sets. I did everything exactly as README says, but didn't get reasonable results. Here are some pictures from the main window when the interactive marker is on:

[image]
[image]
[image]

The metal part fork is the object to be aligned, in the window it seems that the object and point cloud is coincided, but after "run_create_data -d " i get the following picture :
[image] This is the file "0000000001_color_labels" and "000000000x_color_labels" are even worse.

Any advice would be appreciated!

peteflorence, 2019-07-29:
Hi there, the issue is most likely due to the calibration of your camera
and/or the quality of the 3D reconstruction. These items are upstream from
the LabelFusion tool. Sorry we don’t have time to help everyone with their
calibration issues, but good luck.

hygxy, 2019-07-29:
@peteflorence 
Thanks for quick reply. I did use the factory default values for camera intrinsic parameters, I can calibrate the camera manually. But what can I do to improve the quality of 3d reconstruction?

peteflorence, 2019-07-29:
Improving the calibration will improve the 3D reconstruction. Make sure you
use the new calibration parameters in the 3D reconstruction pipeline,
whether it’s ElasticFusion or whatever you are using.

hygxy, 2019-07-30:
@peteflorence 
Hello, I've calibrated my camera manually and get better results now. However, I've also observed another frustrating phenomena: "run_create_data" will create around 600 color images with corresponding labels and poses, some(around 100) of them look good but the rest still presents offsets with different levels and thus can not be applicable. I am wondering if this phenomena is normal because whatever parameters(even "ideal" parameters I suppose) I set in "renderingtrainingimages.py", there always exists offsets in some images, which are quantitatively more than good ones. I am thankful for any ideas!

hygxy, 2019-08-01:
I resolved this issue by changing the intrinsic values in MainController.cpp of ElasticFusion and rebuilding it. Thanks again for your advice! @peteflorence

closed 2019-08-01 by hygxy

Y-pandaman, 2022-03-22:
> I resolved this issue by changing the intrinsic values in MainController.cpp of ElasticFusion and rebuilding it. Thanks again for your advice! @peteflorence

hello ,I use the labelfusion in docker, I have changed intrinsic in the function 'setCameraInstrinsicsAsus()',but not changed the intrinsic values in MainController.cpp of ElasticFusion, the results is bad, in the docker ,how I can do?