RobotLocomotion/LabelFusion #63: libGL issues with nvidia-docker 2.0

andrewqho, 2019-08-20:
We have been trying build the given docker image to run LabelFusion, but we've run into a slew of issues. Mainly, there seems to be an issue with getting libGL to work. We can run the alignment tool, but the scene does not show up in the GUI. We suspect that this has to do with the following issues that appear when we run the alignment tool.

```
libGL error: No matching fbConfigs or visuals found 
libGL error: failed to load driver: swrast
```

A similar error was opened a little over a year last year (link: https://github.com/RobotLocomotion/LabelFusion/issues/25) and the solution was to use nvidia-docker 1.0 instead of 2.0. Unfortunately, docker 1.0 seems to have been deprecated, and that solution doesn't work anymore. 

Has your team - or anybody for that matter - been able to get the LabelFusion image to work with nvidia-docker-2.0?

patmarion, 2019-08-21:
Hi Andrew,

There is a pull request that adds support for nvidia-docker 2.0:

https://github.com/RobotLocomotion/LabelFusion/pull/40

You might try using that branch on @ianre657's fork in order to get nvidia-docker 2 support.  I hope to land a version of that PR but I just haven't had time to work on this project lately.  Good luck!

andrewqho, 2019-08-21:
That solved it, thanks! Many thanks to Mr. Ian Chen :)

closed 2019-08-21 by andrewqho