RobotLocomotion/LabelFusion #64: How to record data?

TrinhTUHH, 2019-09-14:
What should I do after running `lcm-logger`? My log file is unchanged and always 0 KB.

closed 2019-09-15 by TrinhTUHH

khmariem, 2019-09-18:
@TrinhTUHH If you managed to get a log file that is not empty, can you please clarify how you did it? I don't get any error after running `openni2-camera-lcm` and `lcm-logger` so i have no clue so far what might be the reason bhind this issue.
Thanks!

TrinhTUHH, 2019-09-18:
Maybe try installing this package[ rgbd_ros_to_lcm](https://github.com/MobileManipulation/rgbd_ros_to_lcm). Run it before `lcm-logger`
I am using Realsense camera and follow this fork:
https://github.com/ianre657/LabelFusion

khmariem, 2019-09-24:
@TrinhTUHH Thank you for your reply. It didn't work because of a small gaffe from my part and the issue was fixed by running `openni2-camera-lcm` from another terminal that is also executing the container that is already running using `docker exec`.

tiexuedanxin, 2020-05-21:
> @TrinhTUHH Thank you for your reply. It didn't work because of a small gaffe from my part and the issue was fixed by running `openni2-camera-lcm` from another terminal that is also executing the container that is already running using `docker exec`.

I use the prime sense sensor. My log file is unchanged and always 0 KB. Could you give me some advice on how to solve it.

danikhani, 2020-12-24:
> @TrinhTUHH Thank you for your reply. It didn't work because of a small gaffe from my part and the issue was fixed by running `openni2-camera-lcm` from another terminal that is also executing the container that is already running using `docker exec`.

How did you use the exec command? I cannot use the LabelFusion specific bash scripts after opening it in a new terminal with exec command. 
Thank you!

khmariem, 2020-12-24:
1) In a terminal run:  ./docker_run.sh  $path_to_folder
2) Open another terminal and run: docker ps, then check the name of the current running container
3) run: docker exec -it $container_name
4)  run: openni2-camera-lcm

@tiexuedanxin @danikhani

danikhani, 2020-12-24:
>     1. In a terminal run:  ./docker_run.sh  $path_to_folder
> 
>     2. Open another terminal and run: docker ps, then check the name of the current running container
> 
>     3. run: docker exec -it $container_name
> 
>     4. run: openni2-camera-lcm
> 
> 
> @tiexuedanxin @danikhani

Thanks for your reply! The exec command needs two agruments: docker exec [OPTIONS] CONTAINER COMMAND.
So we should give it bash as argument and if so then still the problem of not being able to run the bash scripts is there.

I get `bash: openni2-camera-lcm: command not found` after trying to run openni2-camera-lcm in the second terminal. The first terminal works fine.

danikhani, 2020-12-24:
Ok i solved this problem. Thank you @khmariem for guiding me into the correct direction.

Problem: After running  `docker exec -it CONTAINER bash` the envoirment variables in .bashcr wont be the same as the main terminal. Thats why none of the bash commands like `run_alignment_tool` & `openni2-camera-lcm`  were not working in the new terminal.

Solution:
1. run `set` List the name and value of each shell variable.
2. find the differences using [text-compare.com](text-compare.com) or smiliar sites
3. add the differences to the new terminal opened by exec.
4. run `source .bashrc` to restart the shell for the changes to take effect or source on the new terminal opened by exec.
I tried `run_alignment_tool` & `openni2-camera-lcm`  and both are working in the new terminal.