RobotLocomotion/LabelFusion #67: How to set dimension ofthe mesh file  of own data

hpf9017, 2019-10-08:


hpf9017, 2019-10-12:
thanks to @hoangcuongbk80, the unit is meter, so I change the unit in my model to meter, which is a stl file, the problem is solved.

closed 2019-10-12 by hpf9017

EulerWong, 2020-03-26:
hello, 

> thanks to @hoangcuongbk80, the unit is meter, so I change the unit in my model from mm to m, which is a stl file, the problem is solved.

hello, i meet some problem like this, 

root@cbe969851b41:~/labelfusion/data/logs/2020# run_alignment_tool 
Opening object alignment tool...
log folder: logs/2020
using user specified transform
loading above table pointcloud
showPolyData(colorByName=rgb_colors): array not found
cropping and running ICP
cropping pointcloud to box
computing point to point distance
running ICP
[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.ERROR: In /build/vtk-NWL3SN/vtk-5.10.1+dfsg/Hybrid/vtkIterativeClosestPointTransform.cxx, line 270
vtkIterativeClosestPointTransform (0x5b36050): Can't execute with NULL or empty input

ICP finished

did you meet this problem, i do not know how to fix it. i use a .stl mesh file create using AutoCAD.