RobotLocomotion/LabelFusion #68: use self object-mesh, labels not be fit the rgb image

xiaotmiaomiao, 2019-11-11:
Hi,
 I am trying to make object-mesh by myself,  this objcect-mesh made by point cloud, but obj is not good, please tell me object-mesh make method,  like "drill" is so clear and real well?

I use my intel realsense device to collect point cloude data, next using labelfusion make RGB image and labels data(run_create_data -d), but labels not fit the rgb object，why?  

I think the RGB-D camera must be calibration ，but not sure?  Please tell me, your data(LCMLOG) made method?  

We look forward to your reply，thanks！

origin:
[image]
labels:
[image]
depth:
[image]
pose message:
motor:
  label: 14
  num_pixels: 40234.0
  pose: [[-0.3601003011440027, -0.5683465459219573, 1.742092438769996], [0.04408456497025305, -0.05598430907229312, 0.9378419509410567, 0.3396686375339274]]

closed 2019-11-27 by xiaotmiaomiao

EulerWong, 2020-03-25:
hello, how you make your own mesh, i do not know how to make my own mesh, i download mesh from   internet. do you have any idea to make own mesh?

benghalensis, 2022-04-27:
How was the problem solved?