RobotLocomotion/LabelFusion #74: libGL error: No matching fbConfigs or visuals found libGL error: failed to load driver: swrast

TONG369, 2020-01-02:
ubuntu16.04+cuda8.0
root@a9f15e188b09:~/labelfusion/data/logs/2017-06-13-29# run_alignment_tool Opening object alignment tool...
log folder: logs/2017-06-13-29
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
using user specified transform
loading above table pointcloud
X Error: BadValue (integer parameter out of range for operation) 2
  Extension:    154 (Uknown extension)
  Minor opcode: 3 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 169
  Extension:    154 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x6a00032
X Error: 0 0
  Extension:    154 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 169
  Extension:    154 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x6a00032
X Error: 0 0
  Extension:    154 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 169
  Extension:    154 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x6a00032
X Error: 0 0
  Extension:    154 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 169
  Extension:    154 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x6a00032
X Error: 0 0
  Extension:    154 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 169
  Extension:    154 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x6a00032
X Error: 0 0
  Extension:    154 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 169
  Extension:    154 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x6a00032
X Error: 0 0
  Extension:    154 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 169
  Extension:    154 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x6a00032
X Error: 0 0
  Extension:    154 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 169
  Extension:    154 (Uknown extension)
  Minor opcode: 5 (Unknown request)
[... 254 lines cut ...]
  Extension:    154 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 169
  Extension:    154 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x6a00032
X Error: 0 0
  Extension:    154 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 169
  Extension:    154 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x6a00032
X Error: 0 0
  Extension:    154 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadDrawable 171
  Extension:    154 (Uknown extension)
  Minor opcode: 11 (Unknown request)
  Resource id:  0x6a0000d
Done.
root@a9f15e188b09:~/labelfusion/data/logs/2017-06-13-29#

Jmei1994, 2020-01-09:
Came across the same problem too here.

TONG369, 2020-01-13:
I have solved this problem, you can refer to this link：https://github.com/RobotLocomotion/LabelFusion/issues/63
@Jmei1994

MrLuer, 2020-01-21:
this means the program cannot find the GPU driver，you may not install the nvidia-docer correct,run if 
 the driver  was installed ,run nvidia-smi int docker the out put will the same as in host

EulerWong, 2020-03-04:
you should pull the nvidia-docker2 image, not the nvidia-docker1.

TONG369, 2020-03-30:
@Jmei1994 @MrLuer @EulerWong my prblem is solved.thank you very much!good lucky for you!!!

closed 2020-03-30 by TONG369

tiexuedanxin, 2020-05-20:
> This means the program cannot find the GPU driver，May mean that you did not install the graphics driver properly. if the driver was installed ,run nvidia-smi in docker the out put will the same as in host. I face the same problem before, After I installed nvidia-docker correctly, the driver problem disappeared.

[image]
sorry to bother you, this is my error, could give me some advice to solve the problem, thanks for advance.

MrLuer, 2020-05-20:
@tiexuedanxin 
 it's seems nvidia-docker not installed correctly

tiexuedanxin, 2020-05-20:
> @tiexuedanxin
> it's seems nvidia-docker not installed correctly

i follow the tutorial strictly, could give me some advice on how to solve the problem, thank you for advance.

EulerWong, 2020-05-20:
you can still try this: sudo apt-get install nvidia-docker2

EulerWong, 2020-05-20:
> > @tiexuedanxin
> > it's seems nvidia-docker not installed correctly
> 
> i follow the tutorial strictly, could give me some advice on how to solve the problem, thank you for advance.

you can still try this: sudo apt-get install nvidia-docker2

Starslayerx, 2021-02-04:
I got the same issue:
I have installed nvidia-docker2.5, but my system is Pop!_OS.(Could this be a problem?)
```bash
root@fe7c18c73f81:~# nvidia-smi
Thu Feb  4 10:13:50 2021
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.39       Driver Version: 460.39       CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce MX110       Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   45C    P0    N/A /  N/A |    347MiB /  2004MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
+-----------------------------------------------------------------------------+
```
This is the output:
```bash
root@eb3929d9e245:~/labelfusion/data/logs/2017-06-13-29# run_alignment_tool
Opening object alignment tool...
log folder: logs/2017-06-13-29
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
using user specified transform
loading above table pointcloud
X Error: BadValue (integer parameter out of range for operation) 2
  Extension:    152 (Uknown extension)
  Minor opcode: 3 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 158
  Extension:    152 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x0
X Error: 0 0
  Extension:    152 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 158
  Extension:    152 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x0
X Error: 0 0
  Extension:    152 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 158
  Extension:    152 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x0
X Error: 0 0
  Extension:    152 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 158
  Extension:    152 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x0
X Error: 0 0
  Extension:    152 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 158
[... 280 lines cut ...]
  Extension:    152 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 158
  Extension:    152 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x240000d
X Error: 0 0
  Extension:    152 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 158
  Extension:    152 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x240000d
X Error: 0 0
  Extension:    152 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0
X Error: GLXBadDrawable 160
  Extension:    152 (Uknown extension)
  Minor opcode: 11 (Unknown request)
  Resource id:  0x240000d
Done.
```

welen-zhou, 2021-06-18:
> @Jmei1994 @MrLuer @EulerWong my prblem is solved.thank you very much!good lucky for you!!!

> @Jmei1994 @MrLuer @EulerWong my prblem is solved.thank you very much!good lucky for you!!!

how do you solved the problem. can you give a detail