RobotLocomotion/LabelFusion #78: error: XDG_RUNTIME_DIR not set in the environment when running run_prep

TommasoBendinelli, 2020-01-23:
I get the following when running run_prep: 
root@31dac4aab208:~/labelfusion/data# run_prep
Running pre-processing stage...
rewind is: 0
error: XDG_RUNTIME_DIR not set in the environment.
calling rewind
LcmLogReader::rewind
LcmLogReader found the end of the file.
run done
saving .ply
launch exit
writing pose graph with size 0
deleting eFUsion
writing: ./converted_to_ascii_modified_header.vtp
Done.

A window opens and then immediately close.

TommasoBendinelli, 2020-01-24:
I tried using cuda 10, but then run_trim does not work anymore...

TommasoBendinelli, 2020-01-24:
with Nvidia/cudaopegl it works! If anyone is interested I can share all the procedure

closed 2020-01-24 by TommasoBendinelli

mufengjun260, 2020-05-28:
@TommasoBendinelli Hey,I have the same problem here with 1660ti,could you please share your procedure with me?Thank you!

mufengjun260, 2020-05-28:
I solved it by just deleting the line "set(CUDA_ARCH_BIN "30 35 50 52 61" CACHE STRING "Specify 'real' GPU arch to build binaries for, BIN(PTX) format is supported. Example: 1.3 2.1(1.3) or 13 21(13)")",and it works well on 1660ti!
Thank you anyway!

ctxqlxs, 2021-03-11:
where is the line... @mufengjun260

mufengjun260, 2021-03-11:
In the CMakeLists of ElasticFusion, which link is https://github.com/mp3guy/ElasticFusion/blob/master/Core/src/CMakeLists.txt

ctxqlxs, 2021-03-11:
need to rebuild the docker image? @mufengjun260

mufengjun260, 2021-03-11:
Unfortunately I don’t know that, because I used LabelFusion without Docker.

ctxqlxs, 2021-03-11:
emm, how to use LabelFusion without Docker? Following [this](https://github.com/RobotLocomotion/LabelFusion/blob/master/docs/setup.rst) ?

mufengjun260, 2021-03-11:
It seems like this (the last time I used this repo was a year ago, which caused my confused memory), but I encountered a lot of environmental problems in this way, so I think it is best to use the Dockers image provided by the author

Liuchongpei, 2021-07-03:
@ctxqlxs I solved this problem, you can recompile ElasticFusion in the container after modifying the CMakeLists.txt.

Y-pandaman, 2022-11-03:
> 使用 Nvidia/cudaopegl 它可以工作！如果有人有兴趣，我可以分享所有程序

hello，I try install labelfusion in 2060 super and 3060 laptop，but  it not work, can you share your docker image?