RobotLocomotion/LabelFusion #81: objName is not defined

MrLuer, 2020-03-03:
I want to get bounding box in training images，so I change the branch to pm-render-bounding-box when I run run_create_data got errors below, but in master branch it work well 

Traceback (most recent call last):
  File "/root/labelfusion/scripts/renderTrainingImages.py", line 40, in <module>
    rti = RenderTrainingImages(app.view, app.viewOptions, pathDict)
  File "/root/labelfusion/modules/labelfusion/rendertrainingimages.py", line 34, in __init__
    self.initialize()
  File "/root/labelfusion/modules/labelfusion/rendertrainingimages.py", line 52, in initialize
    self.enableLighting()
  File "/root/labelfusion/modules/labelfusion/rendertrainingimages.py", line 89, in enableLighting
    bbox = obj.findChild(objName + ' bbox')
NameError: global name 'objName' is not defined
Done.

EulerWong, 2020-03-27:
you need to define the var objName where this enableLighting function can access.

EulerWong, 2020-03-27:
replace your enableLighting function with this, as following:

def enableLighting(self):
        view = self.view
        viewOptions = self.viewOptions

        viewOptions.setProperty('Gradient background', False)
        viewOptions.setProperty('Orientation widget', False)
        viewOptions.setProperty('Background color', [0.0,0.0,0.0])

        if view.renderer().GetBackgroundTexture():
            view.renderer().TexturedBackgroundOn()

        for obj in om.findObjectByName('data files').children():
            objName = obj.getProperty('Name')
            obj.actor.GetProperty().LightingOn()
            obj.setProperty('Color', self.storedColors[obj.getProperty('Name')])
            bbox = obj.findChild(objName + ' bbox')
            bbox.setProperty('Visible', self.renderBoundingBox)
        view.forceRender()