RobotLocomotion/LabelFusion #83: penni2-camera-lcm  fails to open realsenseD435i camera

TONG369, 2020-04-01:
Hello, excuse me . I am going to use  Label Fusion to make a dataset. However, this progress has not been successfully. I saw that it was written to support realsense serises cameras. But I run this sentence and it goes wrong. I use an Intel realsenseD435I camera. Do you know how to solve this problem? Please tell me in detail, thank you!!! Good luck!!!
@peteflorence @patmarion

closed 2020-04-13 by TONG369

RobbyHuang, 2020-05-17:
Hi,@TONG369, I met the same problem. Do you know any information to solve it? Could you tell me in detail ? Thanks!

TONG369, 2020-05-20:
@RobbyHuang  You need to run roslaunch in the terminal to open the camera, then execute lcm-logger in the terminal，no in the docker.

RobbyHuang, 2020-05-21:
Thanks,I will have a try.

nihar0602, 2021-04-12:
Hi @TONG369, @RobbyHuang  
I'm also using a similar approach to you and stuck at the same location as you mentioned in this thread. 
Currently, I have a docker image running for this LabelFusion. 
Below are some doubts: 
are you using ROS outside/without docker? If yes, can you elaborate a little bit, I'm confused with that as I'm new to docker. 
I'm also using IntelRealSense D435i. Can you help me with how you made your dataset using LabelFusion? 

Your help is much appreciated. 
Thank You

haijunsu-osu, 2021-06-16:
> Thanks,I will have a try.

did you solve the problem?

haijunsu-osu, 2021-06-16:
> Hi @TONG369, @RobbyHuang
> I'm also using a similar approach to you and stuck at the same location as you mentioned in this thread.
> Currently, I have a docker image running for this LabelFusion.
> Below [...]

any luck with this issue? I have exactly the same issue.