RobotLocomotion/LabelFusion #85: Segmentation fault (core dumped)

TONG369, 2020-04-27:
Excuse me, my operating system is Ubuntu16.04, cuda8.0, and the installation file of the Nvidia driver is installed by .run. When I execute this sentence: run_prep.  I encountered this error, as shown in the following figure:
[image]

When I executed the statement MESA_GL_VERSION_OVERRIDE = 3.3 run_prep, I encountered this error again, as shown below:
[image]
How do you solve this problem? thank you very much!!!

TONG369, 2020-05-14:
the problem is solved

closed 2020-05-14 by TONG369

tom-garford, 2020-06-07:
Hello, how did you solve this issue?