RobotLocomotion/LabelFusion #86: lcmlog-* file's size always  0 bytes.

tiexuedanxin, 2020-05-21:
Hi, I am very interested in your work on LabelFusion. I am trying to create my own data with LabelFusion using  prime sense sensor without a robot.
I installed everything through your Docker: https://hub.docker.com/r/robotlocomotion/labelfusion/.
Then, run: openni2-camera-lcm and lcm-logger as shown in the screenshots below.
However, my lcmlog-* file's size always 0 bytes. Could you give me some suggestions? Thank you very much!
[image]

tsabtahi, 2020-06-08:
Hi @tiexuedanxin  you should not open another separate docker container. Just grab the docker id of the running terminal and on a separate terminal run
sudo docker exec -it <RunningDockerID> bash 
now you need to source the docker startup file so the alias commands can work on the new docker terminal. Use
source /root/labelfusion/docker/docker_startup.sh && /bin/bash

nihar0602, 2021-04-12:
Hi @tsabtahi , 
I'm new to using docker images. 
I have Intel RealSense D435i and i want to create my custom dataset using LabelFusion. 

As the author mentions on using " _First, install librealsense , intel_ros_relasense and rgbd_ros_to_lcm_". The thing that confuses me is, do I need to install these libraries when in a docker container or it should be done on my computer? 

Your help will be much appreciated. 
Thanks!