RobotLocomotion/LabelFusion #87: how to show image in real time when run lcm-logger to collect image?

tiexuedanxin, 2020-05-27:


mufengjun260, 2020-05-28:
Thanks to MobileManipulation/rgbd_ros_to_lcm, I used rviz to show the topic of rgb,and run lcm-logger at the same time.

tiexuedanxin, 2020-05-28:
> Thanks to MobileManipulation/rgbd_ros_to_lcm, I used rviz to show the topic of rgb,and run lcm-logger at the same time.

when i run openni2-camera-lcm and lcm-logger out docker to collect image i can't show the topic of rgb through rviz

mufengjun260, 2020-05-28:
I did it on realsense and don't know how to do that on a openni2 camera.But I think you can try to launch openni2 with ros,and set rgb_topic and depth_topic in rgbd_ros_to_lcm.

freeliuzc, 2020-07-29:
You can use realsense-ros (https://github.com/IntelRealSense/realsense-ros) to publish topics 'A' instead of openni2.  Then use rgbd_ros_to_lcm to subscribe 'A' and publish lcm package 'B'. Last, In one Terminal, run lcm-logger to record package 'B'. In other Terminal, run rviz to subscribe 'A' for visualizing RGB or Point-cloud.