RobotLocomotion/LabelFusion #91: How to run the pipeline at 1280x720 resolution?

Tchaikovic, 2020-06-11:
Hi there, I would like to run the entire pipeline so that the final images have 1280x720 resolution. Everything works fine at VGA resolution. 

When I acquire a HD sequence though, run_prep goes into seg fault. 

Do you know what is required to solve this issue?

peteflorence, 2020-06-11:
Yes, ElasticFusion is compiled to run at VGA resolution:
https://github.com/mp3guy/ElasticFusion/blob/master/GUI/src/MainController.cpp#L37

Won't be able to help with more follow-up, but wanted to point that out.
You can either try recompiling ElasticFusion with your new resolution or
flags to dynamically support different resolutions.

tiexuedanxin, 2020-06-12:
> Hi there, I would like to run the entire pipeline so that the final images have 1280x720 resolution. Everything works fine at VGA resolution.
> 
> When I acquire a HD sequence though, run_prep goes into seg fault.
> 
> Do you know what is required to solve this issue?

hello, do you solve the problem? could you tell me which camera you use and Can you camera collect data in a real time.

Tchaikovic, 2020-06-13:
Hi @tiexuedanxin . Yes, I managed to solve the problem. I'm using both Primesense and Realsense cameras. Follow the instructions here to collect the data https://github.com/ianre657/LabelFusion.

closed 2020-06-25 by Tchaikovic

YangXJ95, 2022-05-12:
Hello @Tchaikovic, do you happen to remember how you solved the problem?

Tchaikovic, 2022-05-12:
Hi @YangXJ95 , I don't remember the details as it was quite some time ago. I can tell you that the resolution is hardcoded across 2-3 files in the repository. I just changed those values consistently and re-compiled. Hope this helps.

YangXJ95, 2022-05-12:
Thank you, this definitely helps! Also as it's hardcoded in the repository, then I am wondering do I need to change anything in the ElasticFusion?