RobotLocomotion/LabelFusion #92: i open the GUI but  nothing is displayed

lzrrr33, 2020-07-23:
when i run  `run_alignment_tool`, i get some errors:
`X Error: BadValue (integer parameter out of range for operation) 2
  Extension:    151 (Uknown extension)
  Minor opcode: 3 (Unknown request)
  Resource id:  0x0
X Error: GLXBadContext 158
  Extension:    151 (Uknown extension)
  Minor opcode: 5 (Unknown request)
  Resource id:  0x0
X Error: 0 0
  Extension:    151 (Uknown extension)
  Minor opcode: 26 (Unknown request)
  Resource id:  0x0`
and the  GUI show nothing(as show below)
recent:///64637ea68dc915b7eae711995f195945
any help would be appreciated!

lzrrr33, 2020-07-24:
Sorry for disturbing you,i have soved this issue

closed 2020-07-24 by lzrrr33

anxiaomi, 2020-12-22:
how did you solve it?

Starslayerx, 2021-02-04:
I got the same issue.